carla-simulator/carla #423: Complex collisions for moving objects (vehicles)

ananiboenig, 2018-05-17:
After creating a scene with moving objects (in this case non player vehicles), we try to capture it with both, Lidar and camera. In the next step, we fused the camera data with the Lidar data. It works perfectly with static objects, however for moving object, the simplification of the contour for moving objects (like the car in the attached image) doesn't correspond anymore to the image view. 

We think that it has something to do with simple collision boxes, that are automatically created for moving objects. Do you know if it is possible to create complex collision boxes for vehicles that matches the actual contours, as seen through the camera? Or are there other ways to detect the actual contour of the moving vehicle with the Lidar?

Thanks! 

[image]

marcgpuig, 2018-05-17:
Hi @ananiboenig,

Yes this is a known issue.
We use raycast to generate the lidar pointcloud, but everything that is "dynamic" uses Unreal's SkeletalMeshes (which can use animations) and this kind of meshes have this issue with the raycast, that use the collision box as collision.
If Unreal change this, it will work fine. If not, we problably could make some image-depth-based-lidar-like sensor that will provide better results :)

Regards.

nsubiron, 2018-05-30:
The problem might be related to the channel used for the ray-cast. We should investigate different channels and ray-cast methods to see if we can get the actual mesh as collision (though that might be slower).

See also the issues reported in #455.

maxjaritz, 2018-11-22:
Hi @nsubiron, @marcgpuig,
Romanenko-Serhii in [this post](https://github.com/carla-simulator/carla/issues/835#issuecomment-440599234) was suggesting a solution. Have you tried that?

marcgpuig, 2018-11-26:
@maxjaritz We didn't try it.
As it looks like it could work for the lidar issue, it may drastically drop the performance because of the hard-to-solve collision. Did you try it?

maxjaritz, 2018-11-26:
@marcgpuig No, I didn't.

germanros1987, 2020-03-19:
We are working on a new LIDAR model as part of the External Sensor Interface, which targets a physically-based laser model. We will follow-up soon.

closed 2020-03-19 by germanros1987

Yaccoub, 2021-10-14:
any news on the physically-based laser model ? thanks