carla-simulator/carla #424: length of episode

Florence-C, 2018-05-17:
Hello, 

I'm trying to run CARLA with client server mode, and each episode is really short. 
How can I set a higher length for the episodes ? 

Thanks, 

Florence

marcgpuig, 2018-05-17:
Hi @Florence-C,

Look at these lines of `client_example.py`:

https://github.com/carla-simulator/carla/blob/c7093b58ad98b4558a8d9a5768f2727cad5df069/PythonClient/client_example.py#L25-L28

Florence-C, 2018-05-17:
Hello @marcgpuig 

Thank you for your response. 
Actually I am using the `make_carla_client` (in carla/client.py) function and there is no such parameter like frame_per_episode. Do you have any idea on how to do so ? 

Thanks 

Florence

marcgpuig, 2018-05-18:
Hello @Florence-C,

I strongly recommend you to try to read and understand all the [`client_example.py`](https://github.com/carla-simulator/carla/blob/master/PythonClient/client_example.py), is a good start example and it is **well commented**.

`make_carla_client()` just opens a connection, in this case named `client`:
```py
with make_carla_client(args.host, args.port) as client:
```
**Inside** that connection we use:
```py
for episode in range(0, number_of_episodes):
    client.start_episode()
    [...]
```
that actually make a loop of episodes.

And **inside** that loop, we have another loop for all the frames:
```py
# Iterate every frame in the episode.
for frame in range(0, frames_per_episode):
```
where we get the information of each frame of the game.

This is just a way to organize our code in episodes and frames. You can do it the way you want :)

Florence-C, 2018-05-18:
Ok thanks a lot !

closed 2018-05-18 by Florence-C