carla-simulator/carla #427: Setting multi-agent in one server for Reinforcement Learning

marintoro, 2018-05-17:
Hello,

I am planning to use Carla to train some RL algorithms. The idea I would like to implement is that in stead of running multiple servers with one agent in each server, I would like to set multiple agent (i.e. cars sending their own camera image and their own information) in the same game instance.
I think that this would be way more computational efficient and allows more agent running on a single computer.
Is that a feature not or currently implemented or is that already possible with some setup needed?

marcgpuig, 2018-05-18:
Hi @marintoro,

Sorry but this is not currently implemented yet.
But hey, good news! @nsubiron just start working on it ;)

Anyway, we need to test if it is really _more computational efficient_, because Unreal is _not designed_ to render a bunch of cameras simultaneously. We'll see :)

marintoro, 2018-05-18:
Thank you for your reply. "way more computational efficient" was maybe a bit strong indeed! ;)

Actually, I implemented this multi-agent settings in Unity (not UE4), each of this agent were invisible and didn't collide with each other. It was speeding up the total frame rate on a single machine by a factor around 4/5 (I was launching only one game instance with around 16 reinforcement learning agents instead of launching 16 game instances each with one agent). 
But it's true that this performance gap was in fact really interesting only with a little number of agent in each game instance (i.e from 1 agent to around 10...). 
Using 1000 agents was really close to be 10 times slower than setting 100 agents for example...

marcgpuig, 2018-05-18:
Well @marintoro it makes a lot of sense.

Thanks for the information and the feedback. We will do these tests when we get a first version of this multi client :)

I close this issue because is already a feature request.

closed 2018-05-18 by marcgpuig