carla-simulator/carla #430: LInux build setup.sh fail

anthonypan08, 2018-05-17:
When I run Setup.sh for Carla on Linux Ubuntu 18.04, it said something about xlocal.h not found. I digged around and it seemed like it might be an issue with Unreal Engine, but my Unreal Engine setup was fine.

[image]

marcgpuig, 2018-05-18:
Hi @anthonypan08,

With a little search I found [this](https://answers.unrealengine.com/questions/705076/xlocaleh-not-found-on-linux.html) question in Unreal forums.
It doesn't looks like a problem in `Setup.sh` :(
Hope it help.

anthonypan08, 2018-05-18:
So did anyone encounter the same problem and have a solution?

18605973470, 2018-05-20:
You can create a soft link by **`ln -s /usr/include/locale.h /usr/include/xlocale.h`** if locale.h exists

marcgpuig, 2018-05-30:
Hey @anthonypan08 
Some good news?
Is the problem solved?

anthonypan08, 2018-05-30:
SOLVED:
The problem was the driver on the Ubuntu machine. I had to change the driver to Nvidia driver instead of using the X.org driver through the "additional drivers" windows.  After that, the Carla worked

closed 2018-05-30 by anthonypan08

J-Rojas, 2018-07-01:
@anthonypan08 That fix doesn't work for me. I have Nvidia drivers installed and enabled and still encounter this compiler bug.

marcgpuig, 2018-07-02:
@J-Rojas are you using the correct version of everything (Unreal 4.18.x and so on)?

J-Rojas, 2018-07-02:
Yes, 4.18.3, latest Clang 3.9.x. I'm using Ubuntu 18.04. The only workaround that worked for me was the symbolic link.