carla-simulator/carla #431: Increasing per-process limit of core file size to infinity.

mhusseinsh, 2018-05-18:
Hello,

I finished the installation successfully by running the Setup.sh file. Now I am trying to run the rebuild file, following this command "UE4_ROOT=~/UnrealEngine_4.18 ./Rebuild.sh"

I receive the following error:

Performing 11 actions (8 in parallel)
[2/11] Compile Module.Carla.cpp
[3/11] Compile Module.Carla.gen.cpp
[1/11] Compile PCH.CarlaUE4.h
[4/11] Compile CarlaUE4GameModeBase.gen.cpp
[5/11] Compile CarlaMapGenerator.cpp
[6/11] Compile CarlaUE4.cpp
[7/11] Compile CarlaUE4GameModeBase.cpp
[8/11] Compile CarlaMapGenerator.gen.cpp
[9/11] Compile CarlaUE4.init.gen.cpp
[10/11] Link libUE4Editor-Carla.so
[11/11] Link libUE4Editor-CarlaUE4.so
Total build time: 72.53 seconds (Local executor: 20.96 seconds)
Launching UE4Editor...
Increasing per-process limit of core file size to infinity.
setrlimit() failed with error 22 (Invalid argument)
- Max per-process value allowed is 0 (we wanted infinity).
Could not adjust core file size, consider changing "core" in /etc/security/limits.conf and relogin.
error(22): Invalid argument
Alternatively, pass -nocore if you are unable or unwilling to do that.
Could not set desired per-process limits, consider changing system limits.
Exiting abnormally (error code: 1)

what do I have to do now  ? @nsubiron 
[image]

nsubiron, 2018-05-22:
I don't know why this would happen, but it seems that passing the `-nocore` argument when launching the editor should work around it. Maybe you can search in the Unreal Engine forums for more info.

mhusseinsh, 2018-05-22:
Hello
thanks for your response.
During the past days, I searched a lot online, and I didn't find definitive answers, but the `-nocore` option was one of the solutions I found.

I tried doing this by adding this flag in the Rebuild.sh file in the command which launches UE4editor, however, this is the error I got afterwards:
[image]

It already started to load the editor, and it reached over 95% then it stopped and I got this above error

mhusseinsh, 2018-05-22:
and when i try to run it by:
~/UnrealEngine_4.18/Engine/Binaries/Linux$ ./UE4Editor -nocore

I receive the following error
[image]

nsubiron, 2018-05-22:
oh it seems there is something wrong with your UE4 installation, maybe you can try reinstalling it. You can find [here](https://wiki.unrealengine.com/Building_On_Linux) more details and troubleshooting for compiling UE4 on Linux.

mhusseinsh, 2018-05-23:
I reinstalled UE4, but I still get the same problem
It is so weird, and I still can't know how to solve it. I am just trying to figure out

closed 2018-07-09 by nsubiron