carla-simulator/carla #432: 3D world coordinates to 2Dimages coordinates of the walkers!

MRAWAY77, 2018-05-19:
Hi I’m using the version 0.8.2 on Ubuntu 16.04. I wanted to ask how can I output the position (x,y) of the walkers in the 720 by 512 sized images when I obtain the location of that walker in the 3D world (x,y,z)?

 What concept does UE4 used in the AsceneCaptureCameraToDisk.cpp for conversing the 3D image in the Simulator into the 2D image output ? Is it tsai camera calibration model ? Or?

marcgpuig, 2018-05-20:
Hi @MRAWAY77,

If you want to project the walkers position to 2D, take a look at issues #56, #64, #314.
The last issue contains [this client](https://gist.github.com/marcgpuig/8317fdb8ffd2a01f7156417d29d05857) where I'm making these calculations.

The process to convert 3D to 2D is made by Unreal, and it is a perspective projection matrix transformation ([info](https://www.scratchapixel.com/lessons/3d-basic-rendering/perspective-and-orthographic-projection-matrix/building-basic-perspective-projection-matrix), and more [info](https://learnopengl.com/Getting-started/Coordinate-Systems)), commonly used in real time rendering.

closed 2018-05-30 by marcgpuig

MRAWAY77, 2018-05-31:
Hi, I managed to get the bounding box and the 2D coordinates but its showing more bounding box then usual. cars that are ain't suppose to be in the FOV of the camera are also displaying their bounding box on screen. Also the bounding box is not referencing to the bounding box extent of the actor but just referencing from the center point. How can i get a bounding box that is always encapsulated the entire car? or display the bounding box based on the box component extent in UE4?

marcgpuig, 2018-05-31:
Hey @MRAWAY77.
How much is this mismatch noticeable?
Looks like an implementation error... others are getting good results like #425.
Make sure the FOV of your Simulator camera and the client FOV (used for the intrinsic matrix) are the same.