carla-simulator/carla #435: Please explain the lidar's point cloud data format. What does (x, y, z) mean?

zhangfree2018, 2018-05-21:


wzhAptiv, 2018-05-22:
Hi @zhangfree2018 , 
The LiDAR data format is now as:
+x is to the front of the car, +y is to the right of the car, +z is to the ground (downwards). 

For other information, I suggest you firstly read the Measurement you can get from the server, which is documented [here](http://carla.readthedocs.io/en/latest/measurements/)

zhangfree2018, 2018-05-24:
Hi @wzhAptiv，
If the format is what you said，what is to the right of the car and to the front of the car? I have plotted the points , what does the value of these points mean? As follows is what I plotted some points:
ply
format ascii 1.0
element vertex 15839
property float32 x
property float32 y
property float32 z
end_header
2.38 -1.72 -0.52
2.38 -1.64 -0.51
2.43 -1.61 -0.51
-13.91 42.22 -7.84
-14.06 39.96 -7.47
-14.61 39.01 -7.34
-15.17 38.15 -7.24
-15.74 37.37 -7.15
-16.16 36.32 -7.01
-16.73 35.65 -6.94
-17.21 34.83 -6.85
-17.58 33.85 -6.73
-18.84 34.56 -6.94

wzhAptiv, 2018-05-24:
Then I really suggest you to plot those data point in some visualization software (quite a lot can visualize ply files). You'll have much better understanding. 
those 'to the direction' means the axis direction it pointing to, please consider in the coordinate system axes.

marcgpuig, 2018-05-24:
Hi @zhangfree2018.

The output you have is correct. These points are **sensor's local space**, so it's up to you to transform the points to world space. Transforming these points be careful with #392.

As @wzhAptiv mentioned, we suggest you to plot these points in some some visualization software like [Meshlab](http://www.meshlab.net/).

closed 2018-05-30 by marcgpuig

zhangfree2018, 2018-05-31:
Hi @marcgpuig @wzhAptiv ,
Thanks for your explanation!