carla-simulator/carla #436: Issue with loading project from Unreal Engine?

anthonypan08, 2018-05-21:
I am not sure if this is an issue with carla or Unreal Engine, but every time I try to open Carla on Unreal Engine, it crashes.
[image]

nsubiron, 2018-05-25:
This is probably an issue with UE4, but it doesn't really show the problem in screenshot you sent. Try opening the editor without the Carla project

    ~/UnrealEngine_4.18/Engine/Binaries/Linux/UE4Editor

You can also check the log files for an earlier error (before signal 11, cause at that point it doesn't give much info).

anthonypan08, 2018-05-25:
I opened the am able to open unreal editor window but if I try to open a new project or existing project, it just crashes. I am using a Linux 18.04 OS

closed 2018-07-09 by nsubiron