carla-simulator/carla #443: Dataset - Images

mhusseinsh, 2018-05-23:
Hello,

I would like to ask if there is by any means a dataset extracted from Carla, which can be downloadable, instead to do that myself using the simulator ? I want to work on images which are extracted from the simulator in different areas.
I already found this [](https://github.com/carla-simulator/imitation-learning)
But the problem, that these are only 200 images, with 200*88 resolution, which is quite low to work on.

Thanks in advance

felipecode, 2018-05-28:
I am sorry @mhusseinsh 
I have only stored the data in this resolution =/. 
My mistake. I eventually will do other datasets, but I will need a diferent
workflow in order to store the images with more resolution.

Cheers mate.

mhusseinsh, 2018-05-28:
Hello @felipecode 
Thanks a lot for your reply.
The problem, that I have some errors while building and getting carla to run, and accordingly, I cannot have another option either to use this dataset. 
But yes, the resolution is very very low to be used in any training and giving out reliable results.
I will try my best, and I will try to get Carla running by any means.

Thanks a lot

felipecode, 2018-05-28:
Well for some problems this resolution is enough. All the results that we had 
uses this resolution http://carla.org/2018/04/23/release-0.8.2/
Of course, if you are working with a different problem, maybe resolution is
an issue.

We will try to help you on your issues.

Cheers.

mhusseinsh, 2018-05-29:
Hello again,
i just wanted to inform about something. I was checking all the h5 files and extracting the images from them. The file 'data_06790.h5', seems to be corrupted. just wanted to point out :)

closed 2018-07-18 by mhusseinsh