carla-simulator/carla #444: CARLA's architecture

JSaffari, 2018-05-24:


JSaffari, 2018-05-24:
Hi CARLA DEVELOPERS
Do you have any use case diagram and carla's architecture to share with me.
We maybe add some feature to carla thus we need your documentation.
I saw your docs bout those were not enough.
Yours truly,JS.

juaxix, 2018-05-24:
Yes @JSaffari, you can get it from CARLA/Docs/img , for example:  carla_plugin_graph.png and carlaserver.svg ,but this is going to change soon!

JSaffari, 2018-05-24:
ERROR: UnrealBuildTool Exception: System.OutOfMemoryException: Exception of type 'System.OutOfMemoryException' was thrown.
Hi @juaxix First of all thanks for your replay.

You say "Change", This is a heavy change?

Our team intend to study CARLA and if possible we change some section to be better to learn amateur drivers and researcher in Uni.
We have a simulation but that developed by OGRE3D and other old SDKs.
We decided to utilize UE4 technology to vary our sim or perhaps CARLA.
Can you help us to process our works? 
Do you have any idea for us?

Tanker.

juaxix, 2018-05-25:
Well, it's part of the [first CodeSprint](https://github.com/carla-simulator/carla/issues?q=is%3Aissue+is%3Aopen+label%3ACodeSprint1 ) we are currently driving now, and check the changes in the clients ( #428 ), networking architecture ( #439 ), windows automatic building ( #448 )
I know OGRE3D, I used it in the past to makes games hehe, I think Unreal may be a little bit advanced hehe
I'm curious about your team, can you tell us more about it and your project? we can talk about the proccess in other issue if you want.
Cheers

JSaffari, 2018-05-31:
Hi dear @juaxix
Sorry, I am late.

First version of our simulator created about 8 years ago, that time we only had simple simulator with simple environment, physical structure( like real car), Vehicle dynamic (Use Physx ) so on, we worked on that about three years we made a simulator that Drivers could use to train, we did not have flexible scenario to better training. It is very hard to create a new scenario because that was hard code and didn't exist any vehicle traffic and shader(for better lighting).
___
Second Version 2.0
We decided to make another software beside simulator to generate scenario simply.
It must be user-friendly thus persons don't have great degree who can use that, We made it and now our trainers are grateful.
Currently we have that software, User or trainer by Scenario Editor can set complex scenario with  vehicles, pedestrians and obstacles.
We have vehicle traffic and night lighting on streets and buildings.
In Current version we have a lot of issue:
-We can't use HMD (Oculus), Ogre could not supports.
-When we use three monitor or more our environment could not load and crashed.
-Most our works is hard code thus our development time is grate that is bad thing for us because our team is small and don't have time to do that.

There are more reason but I must think if you like to know. 
Thanker

marcgpuig, 2018-07-31:
Hi @JSaffari!
As you can see on out master branch these changes have been made.
We recommend you to study CARLA from scratch again!

Thank you so much for your advices, we will consider them :)

Cheers!

closed 2018-07-31 by marcgpuig