carla-simulator/carla #446: How can I get the real time location and categories of objects in the Uworld?

Super-Tree, 2018-05-24:
I want to get the real time location and categories of objects in the Uworld to train my algorithm，but I don't know how to get the information.Is there a function to acquire it?

marcgpuig, 2018-05-24:
Hi @Super-Tree,

Please read the [documentation](http://carla.readthedocs.io/en/latest/) and take a look at the well commented examples that we provide in our repo, like [`client_example.py`](https://github.com/carla-simulator/carla/blob/master/PythonClient/client_example.py).

https://github.com/carla-simulator/carla/blob/998e2421c9dde806e9bef7c5d140238606d267cf/PythonClient/client_example.py#L159

`non_player_agents` contains a list with all this information. On the [Measurements](http://carla.readthedocs.io/en/latest/measurements/) section of our documentation you will find more information.

zhangfree2018, 2018-05-25:
Hi @marcgpuig ,
How can I get player_agents's movement orientation that the simulator displayed on lower left？
 As follows:
Orientation: (x,y,z)

closed 2018-05-28 by Super-Tree

p-schulz, 2018-05-28:
Hi @zhangfree2018 ,
are you looking for something like this?

```
measurements.player_measurements.transform.orientation.x
measurements.player_measurements.transform.orientation.y
measurements.player_measurements.transform.orientation.z
```
It's marked as deprecated, while instead of orientation (Vector3D) the rotation (Rotation3D, pitch/roll/yaw) should be used. You can simply print it to the screen/console (like in client_example.py) or dump it to a .txt file:

```
message = 'Orientation: ({rot_x:.1f}, {rot_y:.1f}, {rot_z:.1f})'
message = message.format(
    rot_x=measurements.player_measurements.transform.orientation.x,
    rot_y=measurements.player_measurements.transform.orientation.y,
    rot_z=measurements.player_measurements.transform.orientation.z)
with open("_out/orientation.txt", "a") as text_file:
    text_file.write(message)
    text_file.close()
```

I do this inside the print_measurements() function of the client_example.py. I hope this is helping you.

zhangfree2018, 2018-05-29:
Hi @p-schulz ,
Thanks for your help! I have got orientation by your explanation.