carla-simulator/carla #448: Automatic build system for Windows

marcgpuig, 2018-05-24:
## Introduction
There are more people than expected using CARLA in Windows :)
We currently have the [documentation](http://carla.readthedocs.io/en/latest/how_to_build_on_windows/) on how to **manually** build CARLA in Windows platforms.
Also you can find a lot of information and support in the issue #21.
But what we really want is an automatized installation right?

## What do we want these scripts to do?
- [x] Download and install Protobuf 3.3.2.
- [x] Download and install Boost 1.64.
- [ ] Download and install Googletest.
- [ ] Download and unzip the right version of the content (assets) from Google Drive.
- [x] Clean intermediates and recompile carlaserver.

All those features must try to:
- Check for missing dependencies.
- Constantly check for errors.
- If errors, notify it and try to give a solution to the user.

## Final user case
#### Required software:
- Git
- Make
- Cmake
> Be sure that these three programs are added to your path, so you can use them from your command prompt.
- Unreal Engine (v4.18.x)
- Visual Studio (2017 preferably)

Run:

    $ git clone https://github.com/carla-simulator/carla.git
    $ cd carla
    $ Setup.cmd -j 8 & Rebuild.cmd

It will open a Carla Unreal project ready to be modified as you want!

nsubiron, 2018-05-28:
Don't forget to copy

    Docs/Example.CarlaSettings.ini --> Unreal/CarlaUE4/Config/CarlaSettings.ini

marcgpuig, 2018-05-28:
Right!