carla-simulator/carla #450: Abnormal Driving Dataset in Autopilot Mode

tayloryang722, 2018-05-25:
Hi CARLA team,

I have a question about collecting the abnormal driving dataset.

My original plan is to collect a few datasets in the autopilot mode that doing everything perfectly and collect a few datasets that drive overspeed and doing unstable steering motion in the manual control mode.

I personally experimented in collecting datasets in the manual mode and run into the issue of performance limitation of pygame that prevents me from driving. The python client becomes extremely sluggish when saving images and the pygame cannot take prompt keyboard inputs anymore so that the car can only speed up to 2km/hr no matter how I press the keyboard.

From that experience, I figure another way to collect abnormal data. I can increase the range of the steer to (-0.65,0.65) to make car driving behavior differently a lot from the regular driving in the autopilot mode. As I expected, I can see the car drive in the opposite lane or sidewalk etc. However, after a few seconds, the car driving into the neighborhood and is stuck in a house.

**1.My first question is that what is your suggestion or advice to collect abnormal data efficiently in either manual control or in autopilot mode with a larger range of noise? For the autopilot mode with larger range of noise solution, I expect car to driving longer so that I can collect a larger dataset like 3mins?**

Another thing I try out to get the abnormal dataset in the autopilot mode is to increase the throttle to make car drive overspeed. I basically time the throttle value 2 times and send them back to the controller. What I observed now is that the car will speed up faster than before. In the 30km/h speed limit zone, it can drive up to 32 km/h sometimes but in 90km/h speed limit zone, it will still follow the rule of speed limit. 

**2. My second question is that is there any way to make car drive overspeed in the autopilot mode.**

My takeaway for now is that due to the performance limitation of the pygame, the user cannot collect datasets in the manual mod as they wanted ( in my case is to collect datasets driving crazy and overspeed). I think adjust the input from the autopilot control is a more promising solution, but I am stuck in what is the best way to achieve this goal.

Many thanks in advance.

Best regards,
Taylor

felipecode, 2018-05-28:
Hello @tayloryang722 
Thanks for the issue. I will try to answer your questions.

1. Pygame is slow: Indeed, it is very hard to drive with pygame. One
turnaround is setting the outputscreen resolution very small and reducing
the resolution of your screen so you can se it better. However this 
does not solve the problem when there are several sensors.
We are currently working on that on issue #421.
Also considering using a joystick for driving. Driving on the keyboard is very
unpractical for any serious data collection. Check issue #88 

2. The autopilot has limitations, it lacks flexibility since it always obeys the speed
limits of the zone. Also, its fixed policy does not allow for you to build a more robust
recovery from noise for instance. We are currently on a code sprint to have a robust
client side agent #428 that will hopefully enable you to easily adapt for your needs !
If it interests you , you can eventually also contribute to our code sprint. Consider joining our discord chat for a more instant discussion on that https://discordapp.com/invite/vNVHXfb

Cheers,
Felipe.

closed 2018-07-09 by nsubiron

AOOOOOA, 2024-05-16:
> 嗨卡拉团队，
> 
> 我有一个关于收集异常驾驶数据集的问题。
> 
> 我最初的计划是收集一些在自动驾驶模式下一切都完美的数据集，并收集一些在手动控制模式下超速行驶和转向不稳定的数据集。
> 
> 我亲自尝试以手动模式收集数据集，并遇到了 pygame 性能限制的问题，导致我无法驾驶。 python客户端在保存图像时变得非常缓慢，并且pygame无法再及时接受键盘输入，因此无论我如何按下键盘，汽车只能加速到2公里/小时。
> 
> 根据这次经验，我想出了另一种收集异常数据的方法。我可以将转向范围增加到（-0.65,0.65），以使汽车驾驶行为与自动驾驶模式下的常规驾驶有很大不同。正如我所料，我可以看到汽车在对面车道或人行道等处行驶。然而，几秒钟后，汽车驶入附近并被困在一所房子里。
> 
> **1.我的第一个问题是，您对在手动控制或噪声范围较大的自动驾驶模式下有效收集异常数据有何建议或建议？对于具有更大范围噪声解决方案的自动驾驶模式，我希望汽车行驶更长时间，以便我可以收集更大的数据集（例如 3 分钟）？**
> 
> 我尝试在自动驾驶模式下获取异常数据集的另一件事是加大油门以使汽车行驶超速。我基本上对油门值进行两次计时并将它们发送回控制器。我现在观察到车子会比以前加速得更快。在30公里/小时限速区，有时可以行驶到32公里/小时，但在90公里/小时限速区，仍然会遵循限速规则。
> 
> **2.我的第二个问题是有什么办法可以让汽车在自动驾驶模式下超速行驶。**
> 
> 我现在的结论是，由于 pygame 的性能限制，用户无法按照自己的意愿在手动模式中收集数据集（在我的例子中是收集驾驶疯狂和超速的数据集）。我认为调整自动驾驶仪控制的输入是一个更有前途的解决方案，但我陷入了实现这一目标的最佳方式。
> 
> 提前谢谢了。
> 
> 最好的问候， 泰勒

Hi Have you build the abnormal driving dataset? If so, could you please share a link with me? I would appreciate that.