carla-simulator/carla #452: the question of How to run the client.example.py

errorJY, 2018-05-26:
when i run the client.ecample.py.I meet the ImportError :No module named 'google'.It's all:
`Traceback (most recent call last):
  File "/home/error/Carla_0.8.2/PythonClient/carla/transform.py", line 18, in <module>
    from . import carla_server_pb2 as carla_protocol
  File "/home/error/Carla_0.8.2/PythonClient/carla/carla_server_pb2.py", line 6, in <module>
    from google.protobuf import descriptor as _descriptor
ImportError: No module named 'google'
But i had make the protobuf.`when i  input it:
$ protoc --version
i will get libprotoc 3.5.1.  
so i don't know how to do with it.
Thank you for your  care.

nsubiron, 2018-05-28:
Hi @errorJY,

Is it possible that you don't have the dependencies installed in Python3?

    sudo pip3 install -r PythonClient/requirements.txt

Which Python version are you using for running `client_example.py`?

errorJY, 2018-05-31:
Hi @nsubiron 
when i installed the dependencies,i think  it will more fast:
`sudo pip3 install -i https://pypi.tuna.tsinghua.edu.cn/simple -r PythonClient/requirements.txt`
It's will more fast in China...for the wall.  u can  write it when u upgrate  the doc.

abhishekrana, 2020-03-05:
**SOLUTION:**
pip install protobuf

**Issue logs:**
(carla-0.8.2_py3.5.5) $ python client_example.py --carla-settings=../Example.CarlaSettings.ini --autopilot
`
Traceback (most recent call last):
`
`
  File "/data/user/git/carla-0.8.2/PythonClient/carla/transform.py", line 18, in <module>
    from . import carla_server_pb2 as carla_protocol
  File "/data/user/git/carla-0.8.2/PythonClient/carla/carla_server_pb2.py", line 6, in <module>
    from google.protobuf import descriptor as _descriptor
`
`
ImportError: No module named 'google'
`
`
During handling of the above exception, another exception occurred:
`
`
Traceback (most recent call last):
`
`
  File "client_example.py", line 18, in <module>
    from carla.client import make_carla_client
  File "/data/user/git/carla-0.8.2/PythonClient/carla/client.py", line 14, in <module>
    from . import sensor
  File "/data/user/git/carla-0.8.2/PythonClient/carla/sensor.py", line 19, in <module>
    from .transform import Transform, Translation, Rotation, Scale
  File "/data/user/git/carla-0.8.2/PythonClient/carla/transform.py", line 20, in <module>
    raise RuntimeError('cannot import "carla_server_pb2.py", run '
`
`
RuntimeError: cannot import "carla_server_pb2.py", run the protobuf compiler to generate this file
`

**Solution logs:**
(carla-0.8.2_py3.5.5) $ pip install protobuf
`
Collecting protobuf
  Downloading protobuf-3.11.3-cp35-cp35m-manylinux1_x86_64.whl (1.3 MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 1.3 MB 171 kB/s 
`
`
Requirement already satisfied: six>=1.9 in /home/user/.local/lib/python3.5/site-packages (from protobuf) (1.12.0)
`
`
Requirement already satisfied: setuptools in /home/user/.conda/envs/carla-0.8.2_py3.5.5/lib/python3.5/site-packages (from protobuf) (40.2.0)
`
`
Installing collected packages: protobuf
`
`
Successfully installed protobuf-3.11.3
`

(carla-0.8.2_py3.5.5) $ python client_example.py --carla-settings=../Example.CarlaSettings.ini --autopilot
`
INFO: listening to server localhost:2000
`
`
CarlaClient connected
`
`
Starting new episode...
`
`
Vehicle at (339.0, 218.2), 3 km/h, Collision: {vehicles=0, pedestrians=0, other=0}, 0% other lane, 0% off-road, (0 non-player agents in the scene)    ^C
Cancelled by user. Bye!
`

**Additional info:**
(carla-0.8.2_py3.5.5) $ python --version
`
Python 3.5.5 :: Anaconda, Inc.
`

(carla-0.8.2_py3.5.5) $ pip --version
`
pip 20.0.2 from /home/user/.conda/envs/carla-0.8.2_py3.5.5/lib/python3.5/site-packages/pip (python 3.5)
`

[requirements.txt](https://github.com/carla-simulator/carla/files/4292634/requirements.txt)