carla-simulator/carla #453: How to get a precise point cloud of the non-players?

YlugeR, 2018-05-28:
Dear Carla Developers
     Through lidar simulation, I got the cloud-point models of non-players in Carla. However, I found that the models in point cloud are too simple, the vehicles are simplified into rectangles, and pedestrians are simplified into cylinders. This simplified model is not suitable for training samples. I want to know how to get the accurate non-player point-cloud models.
Thank you!

marcgpuig, 2018-05-28:
Hi @YlugeR.
This is a duplicated question #423.

closed 2018-05-28 by marcgpuig

YlugeR, 2018-05-28:
Dear @Marc Garcia Puig
I'm sorry for my duplicated question.In addtion, I have another question: how could I get the reflected intensity of lidar in the carla?

marcgpuig, 2018-05-28:
Also another duplicated :) #327