carla-simulator/carla #455: Cannot get RoadLines label from a LiDAR sensor

EhsanCode, 2018-05-29:
I am trying to get the object class for LiDAR. I have modified ALidar::ShootLaser and I have added this code:
`
uint8_t ObjectLabel  = static_cast<std::underlying_type<ECityObjectLabel>::type>(ATagger::GetTagOfTaggedComponent(*HitInfo.GetComponent()));
`

But it seems there is a problem to get Vehicles and RoadLines labels. By tracking the issue I realized  
`
 GetWorld()->LineTraceSingleByChannel(
    HitInfo,
    LidarBodyLoc,
    EndTrace,
    ECC_MAX,
    TraceParams,
    FCollisionResponseParams::DefaultResponseParam
  );
`
never hit RoadLines and HitInfo never contains any component of  Vehicles nor RoadLines.

In this visualization points are colored based on their labals, you can see there is no **RoadLine** label:

[image]

But I could get this information from a  'SemanticSegmentation' camera.  

[image]

Thank you so much,
Best regards,
Ehsan

nsubiron, 2018-05-30:
Hi @EhsanCode,

The ray-cast used by the lidar uses the collision channel, so only things with collision enabled will appear. Things like road-lines do not have collision for performance reasons. See also this issue with the vehicles #423. In the case of the vehicle the collision occurs with their physics asset instead of the mesh, I'm guessing that's the reason why they don't have a tag.

EhsanCode, 2018-06-01:
Hi @nsubiron 
Thank you so much for your attention. I've enabled the collision on every roadLine content. Now, they have the same collision configuration as shown in this figure:
[image]

but still, no roadLine collision has been detected. Would you please let me know if I could do anything more to detect roadLine collision?
Thanks again.

nsubiron, 2018-06-01:
Hi @EhsanCode,

Hmm I see the problem. Even if you set collision to BlockAll, the road lines do not have a collision mesh. You could add it manually but you'd have to do so for each road line, doesn't seem like a very optimal solution.

I'm suspecting we can change the lidar ray-cast configuration to go on the visibility channel, it is probably slower but it should hit the right item every time. If this works, it would fix the issues with the dynamic agents too. I'll try to get some time next week to experiment with this.

Btw, we were talking about different ray-cast options in the #cpp channel of our [Discord](https://discord.gg/vNVHXfb).

EhsanCode, 2018-06-08:
Hi @nsubiron 
Thank you so much for consideration.
As you mentioned, It seems collision mesh is not able to present some material such as fences. For example, as shown in the following image:

[image]

fence structure is not detectable with this method. I do not know if it is possible on the visibility channel?

Thank you so much
Best regards,
Ehsan

closed 2018-08-14 by stale[bot]

xuwenqian66, 2018-12-07:
> Hi @nsubiron
> Thank you so much for your attention. I've enabled the collision on every roadLine content. Now, they have the same collision configuration as shown in this figure:
> [image]
>
> but still, [...]

> I am trying to get the object class for LiDAR. I have modified ALidar::ShootLaser and I have added this code:
> `uint8_t ObjectLabel = [...]

Hello, I'm a beginner for this software.
May I ask you, how did you plot this point cloud out? How could you combine the UE4 with the lidar?

marcgpuig, 2020-07-31:
Road lanes are now detected by the new LiDAR introduced in #3131.