carla-simulator/carla #456: Makefile:580: recipe for target 'CarlaUE4Editor' failed

adaxidedakaonang, 2018-05-29:
When I run `sudo UE4_ROOT=~/UnrealEngine_4.18 ./Rebuild.sh`,I met this question:
`Build CarlaUE4 project...
mono "/home/server109/UnrealEngine_4.18/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/server109/ADAXI/carla-0.8.0/Unreal/CarlaUE4/CarlaUE4.uproject" 
Using 'git status' to determine working set for adaptive non-unity build.
Creating makefile for CarlaUE4Editor (no existing makefile)
Performing full C++ include scan (no include cache file)
Using clang version '3.9.1' (string), 3 (major), 9 (minor), 1 (patch)
Building UnrealHeaderTool...
Using 'git status' to determine working set for adaptive non-unity build.
Creating makefile for UnrealHeaderTool (no existing makefile)
Using clang version '3.9.1' (string), 3 (major), 9 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using fast way to relink  circularly dependent libraries (no FixDeps).
Target is up to date
Total build time: 1.45 seconds (NoActionsToExecute executor: 0.00 seconds)
Parsing headers for CarlaUE4Editor
  Running UnrealHeaderTool "/home/server109/ADAXI/carla-0.8.0/Unreal/CarlaUE4/CarlaUE4.uproject" "/home/server109/ADAXI/carla-0.8.0/Unreal/CarlaUE4/Intermediate/Build/Linux/B4D820EA/CarlaUE4Editor/Development/CarlaUE4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
Refusing to run with the root privileges.
Error: UnrealHeaderTool failed for target 'CarlaUE4Editor' (platform: Linux, module info: /home/server109/ADAXI/carla-0.8.0/Unreal/CarlaUE4/Intermediate/Build/Linux/B4D820EA/CarlaUE4Editor/Development/CarlaUE4Editor.uhtmanifest, exit code: Canceled (1)).
Makefile:580: recipe for target 'CarlaUE4Editor' failed
make: *** [CarlaUE4Editor] Error 1
`
my UnrealEngine version is 4.18, and I try 
`
$ cd Unreal/CarlaUE4/ 
$ make CarlaUE4Editor ARGS=-clean 
$ make CarlaUE4Editor
`
I still have this question. Should I update carla? My version is carla-0.8.0

nsubiron, 2018-05-30:
It seems that Unreal doesn't want to be run with `sudo`

> Refusing to run with the root privileges.

adaxidedakaonang, 2018-05-30:
@nsubiron But if I don't use sudo will get 'Permission denied' error:
`
Cleaning up CARLA Plugin...
Cleaning up CARLAUE4...
rm: cannot remove 'Intermediate/Build/BuildRules/CarlaUE4ModuleRules.dll': Permission denied
rm: cannot remove 'Intermediate/Build/BuildRules/CarlaUE4ModuleRulesSourceFiles.txt': Permission denied
rm: cannot remove 'Intermediate/Build/BuildRules/CarlaUE4ModuleRules.dll.mdb': Permission denied
rm: cannot remove 'Intermediate/Build/Linux/CarlaUE4Editor/DependencyCache.bin': Permission denied
rm: cannot remove 'Intermediate/Build/Linux/B4D820EA/CarlaUE4Editor/Development/CarlaUE4Editor.uhtmanifest': Permission denied

`

nsubiron, 2018-05-30:
That's probably the result of a previous run with sudo :/ the easiest probably is to remove all these build folders manually and rebuild Carla following the instructions in the documentation.

adaxidedakaonang, 2018-06-01:
OK，thank you！I will try it.

closed 2018-07-09 by nsubiron

adaxidedakaonang, 2018-10-23:
> Excuse me, is your problem solved? I have the same problem as you.

Hi, sorry that I don't try it,  你好，我没有尝试因为我没有足够的流量

sujie001, 2018-10-23:
> > Excuse me, is your problem solved? I have the same problem as you.
> 
> Hi, sorry that I don't try it, 你好，我没有尝试因为我没有足够的流量

Thank you for your reply！

Unlingius, 2018-12-20:
I faced with the same problem. If I launched with sudo, i got error writing above. When without - i got error permission denied about file user-config.jam. I solved it with the command (before any building operations):
sudo chown <username> user-config.jam
And launch without sudo according to instructions and now it works.