carla-simulator/carla #457: Can I accelerate simulation clock speed?

kanchi0914, 2018-05-29:
Hello,

I want to use CARLA as environment of reinforcement learning.
But I think it will take so long time for learning if simulation clock is same as real-time.
So I'm looking for a way to accelerate simulation clock speed, is there any way?
I remember your team applied RL to CARLA in your paper and hope to know how did you deal with this point.

Thank you for your help.

wzhAptiv, 2018-05-29:
use `-benchmark and -fps=[fps]`, 
then the simulation will go to the next frame whenever the server receives a control package. 

e.g. you set -fps=20, and you send the control package 40 times per second. You are actually having a real-time world running in twice of the speed. 

see [Fixed time-step](http://carla.readthedocs.io/en/latest/configuring_the_simulation/) section for more information.

kanchi0914, 2018-05-30:
Thank you for your quick reply! I'm sorry for overlooking it.
Incidentally, do you have any ideas about recommended or maximum FPS in learning?(on the presupposition that we have sufficient PC)

wzhAptiv, 2018-05-30:
@kanchi0914 the benchmark fps only affects the engine time ticks, so you can set it to be whatever you need in your task. The actually runtime is only affected by the PC hardware/environment. Since you're planning to do reinforcement learning, I think it also applies on any actually runtime if you won't do any HCI. 

I don't see any limitation or restriction on FPS in my opinion. you're free to choose any fps you want.

marcgpuig, 2018-05-30:
Closing the issue because seems resolved.
You can keep talking anyway :)

Thanks @wzhAptiv.

closed 2018-05-30 by marcgpuig