carla-simulator/carla #460: Howto create a DLC for CARLA

juaxix, 2018-05-30:
If you want to create additional content for CARLA, you have to create a plugin with downloadable content. 

This issue is designed to add more info about that particular process.

We will be using this [link](http://www.tomlooman.com/add-mod-support-to-your-unreal-engine-4-game/) as reference.

Meanwhile, to know more about how to make contents for CARLA, go to 
http://carla.readthedocs.io.

child of #438

closed 2018-06-04 by juaxix

juaxix, 2018-06-06:
If you don't want to modify the Unreal Engine source code, there is another way, using patching.

These are the steps to do it.
1) Create ProjectLauncher profile to build the game in full (without additional assets) - FullGame
2) Do the cook of assets, build the game for Windows (or other platform), use the projeclauncher profile launch button for FullGame profile
3) Create additional assets now, cook them and create a projectlauncher profile for patching - Patch
4) launch the patch projectlauncher profile
5) copy the new patch pak _P (patch) in the FullGame build CarlaUE4/Content/Paks
6) the CARLA build in FullGame build dir will load the new assets from this pak 

FullGame profile:

[image]

where your default map will be Town01 and Town02

[image]

Patch profile:

[image]

[image]

KevinYuk, 2018-08-07:
How long does it take to build LinuxNoEditor release version?