carla-simulator/carla #462: change camera to save images

Florence-C, 2018-05-30:
Hello, 

I would like to save images from outside of the car (like in the demo of Carla) how can I do it ? So far I can only save images from inside the car by setting the parameter `save_image=True`. 

Thanks !

marcgpuig, 2018-05-30:
Hi @Florence-C.

You will have to modify the **CARLA Settings** ([documentation](http://carla.readthedocs.io/en/latest/carla_settings/)).

You can do it from 2 different ways:
- **CarlaSettings.ini**: Modifying these lines (Also notice you can add a rotation too):
https://github.com/carla-simulator/carla/blob/3fc39e835de4a3bf036dd80023789e0041a89982/Docs/Example.CarlaSettings.ini#L78-L85

- **Your Python client**: Take a look at the `PythonClient/client_example.py` where you will be able to read the comments on how CARLA clients work. In your case you are searching for these lines, where the settings are being created:
https://github.com/carla-simulator/carla/blob/3fc39e835de4a3bf036dd80023789e0041a89982/PythonClient/client_example.py#L64-L65

Florence-C, 2018-05-31:
Ok great thank you very much !

closed 2018-05-31 by Florence-C