carla-simulator/carla #463: Benchmarking question

eds89, 2018-05-30:
Hello,
Does the trained models (imitation learning, reinforcement learning, modular pipeline, etc) react to changes in the environment (input data)? For instance, if I extend an existing weather condition or create a post process, designed to add dust noise to the camera, will it impact on task benchmarking scores, when running it against existing trained data? I imagine that yes, but I'd like to double-check.

Kind regards,

mvpcom, 2018-05-31:
@eds89 Generally, neural networks and machine learning models cannot generalize very well on entirely different and new sets. The train and test set should be from a similar distribution for an excellent performance. If you check the [results](http://carla.org/2018/04/23/release-0.8.2/), you can see for new town and weather, the performance of the model dramatically declined. So, you can expect extended a weather condition or created a post process will impact on task benchmarking scores.

eds89, 2018-06-01:
> This is what I imagined -- and is what I want to actually do.

Hi,
I wonder if the benchmarks are carried out with the actual camera data, as
the car navigates on the road, or with other game-provided data, e.g.
distance to obstacles, without making use of the captured image data? I'm
probably wrong on that one, but I'd like to double check.

eds89, 2018-06-06:
Up!

closed 2018-08-12 by stale[bot]