carla-simulator/carla #465: Implement UDP server

nsubiron, 2018-05-31:
Child of #439.

First investigate if there is an open-source library out there that we can easily use.

If none of them matches our needs, (we need something pretty simple) implement one. Ideally we should have a similar interface to our TCP server so we can switch between both if needed.

It would be great to find ways of integrating this with rpclib.

nsubiron, 2018-06-11:
I've been experimenting a bit and have a comparison of streaming images through different methods (measured in FPS)

image size |  raw udp   |  raw tcp   | rpclib  
-----------|------------|------------|--------
1920x1080  | 400 - 1.2k | 400 - 700  | ~170
  800x600  |    3k - 6k |       ~4k  | ~750
  100x100  |      +600k |     ~150k  | ~17k

Both "raw" UDP and TCP using Boost.Asio.

Although UDP looks good, the complexity of implementing a good communication is huge compared to the other methods, and depends a lot on low level configuration, like size of the packets. With those frame-rates the client is already too slow to catch all the packets and many get lost, so we need extra code for reliability. It is very easy to end up being slower than TCP.

To translate to Carla units, stereo setup with ground truth, 6 cameras at 90 FPS = 540 FPS. I think we still have margin to keep using TCP for a while.

germanros1987, 2018-06-11:
Uhm. RPC does not look good.

nsubiron, 2018-06-11:
No indeed, but my idea is to use RPC for the "control" of the scene (what's now the world thread in the server), all small messages; and for now we'll keep sensors in a separate thread streaming with TCP (now measurements thread). Later we can decide if it's worth it implementing UDP, but at it is now the network is still faster than the render.

closed 2018-06-13 by nsubiron

analog-cbarber, 2018-07-06:
If the client is on the same machine, the fastest way to stream sensor data would be to use shared memory or use an OS-specific memory transfer technique over a pipe or message queue.

analog-cbarber, 2018-07-06:
Another thing that would probably work even for the remote connection case would be to use fast compression (e.g. Blosc library) on the sensor data. This could be a significant win.