carla-simulator/carla #467: Set the maximum speed for vehicles

lockeregg, 2018-06-04:
Can I set the maximum speed for both player and non-player vehicles?

Thank you!

nsubiron, 2018-06-07:
Hi @lockeregg,

The only way to achieve this at the moment is to get Carla source, modify it, and compile it. You could either modify the speed-limit signs that are in the map from the UE4 editor, or if you want to modify the vehicle's behavior, adapt the C++ code in "WheeledVehicleAIController.cpp".

lockeregg, 2018-06-22:
Hi @nsubiron 
When I set the player throttle = 0.5f, the maximum speed will be 20Km/hr.
Can I increase the maximum speed?

Thank you!

closed 2018-07-09 by nsubiron

queen97king2019, 2020-07-06:
hi, 
Were you able to solve this issue?
I am trying to take the speed as input from keyboard. How do i make the car movements accordingly?