carla-simulator/carla #470: What is the order of lidar's points data?

zhangfree2018, 2018-06-04:
I have saved the lidar's points, but I don't understand that the order of the points data. I want to confrim that whether these points are arranged according to the order of the channels. I guessed that the data were arranged  for the first channel first , then the second channel, the third and so on. And I can get the number of point clouds per channel . As follows:
ply
format ascii 1.0
element vertex 32460
property float32 x
property float32 y
property float32 z
end_header
4.74 -39.08 -6.94
5.10 -35.99 -6.41
5.10 -31.43 -5.61
5.38 -29.43 -5.28
6.00 -29.43 -5.30
6.62 -29.43 -5.32
................................
-0.66 -6.58 3.82
-0.53 -6.60 3.82
-0.40 -6.61 3.82
-0.27 -6.61 3.82
-0.13 -6.62 3.82

marcgpuig, 2018-06-04:
Hi @zhangfree2018.
Maybe this helps?
https://github.com/carla-simulator/carla/blob/4c5fe773d5b93f097e907f3998420ba814f9bbf1/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/LidarMeasurement.h#L13-L31

zhangfree2018, 2018-06-04:
Hi @marcgpuig ,
Thank you very much for your prompt response. I have plotted these points to verify the order of these points.

marcgpuig, 2018-06-07:
Good!

closed 2018-06-07 by marcgpuig