carla-simulator/carla #471: Simulate specific Lidar geometrics

vvaquero, 2018-06-04:
Hi, 
Is it be possible to accurately define specific Lidar geometrics? At the moment I see we can define the number of channels, the upper and lower FOVs, and the total number of points, but these variables do not accurately describe the geometric model of a sensor.

In example, I want to simulate a Velodyne HDL64 sensor, and for that I would need to set a specific horizontal step of the lidar rotation. In addition, for this sensor the separation between the lower layers is different from the upper ones, so it would be useful to introduce a vector with the vertical FOV of each layer.

nsubiron, 2018-06-07:
Hi @vvaquero,

At the moment this is not possible, you could either try what @wzhAptiv suggested, or modify the "Lidar.cpp" to implement a different architecture.

vvaquero, 2018-06-08:
Ok, its a good idea to try with two sensors. Thanks!

closed 2018-07-09 by nsubiron