carla-simulator/carla #474: Question about Fixed time-step

zhangfree2018, 2018-06-06:
I don't understand that the docs explained the question about ' Fixed time-step'. I will try to describe a phenomenon by setting ' Fixed time-step '.  I found that the larger the parameter（-fps=5, 10,20 and so on） setting, the slower the agents run in the simulator. My understanding of the document is that the larger the parameter(-fps= **), the faster the movement is. But the result is exactly the opposite. Is my understanding wrong?

marcgpuig, 2018-06-06:
Hey @zhangfree2018.
I have to tell you to read the [issue related docs](http://carla.readthedocs.io/en/latest/configuring_the_simulation/#fixed-time-step) carefully, but anyway...

### Let's pose an experiment:
Let's say you are going to play 2 simulations and save each image to disk on two different folders **A** and **B**.

1. First you record with `-fps=60` and stores the images to folder **A**.
   - While recording you will see the simulation going "slowly".
   - Looking the recorded images, the elapsed simulation time between images is small.
1. Then you record with `-fps=5` and stores the images to folder **B**.
   - While recording you will see the simulation going "fast".
   - Looking the recorded images, the elapsed simulation time between images is bigger.

1. Now we want to play 2 videos with all the consecutive images:
   - Video with all the images of folder **A**:
We should play the video at **60 frames per second** so 1 second in the simulation is 1 second in real life.
   - Video with all the images of folder **B**:
We should play the video at **5 frames per second** so 1 second in the simulation is 1 second in real life.

---

Your machine will try to generate te maximum number of individual frames each second, regardless the `-fps` parameter. What `-fps` does is determine the **simulation time** between each frame. This generates this "_fast/slow effect_".

Hope it helps.
Regards.

zhangfree2018, 2018-06-06:
Hi  @marcgpuig
Thanks for your detailed explanation about my confusion. Your example can explain the problem clearly.