carla-simulator/carla #485: Low performance on windows

habibutsu, 2018-06-08:
In trying to run CARLA with flag `-carla-server` the performance is dramatically slow down, FPS = 3. In the same time when running CARLA without this flag everything works fine.

Environment:
 * Windows 10 Pro, 16Gb RAM
 * Geforce GTX 770
 * CARLA release 0.8.2 (Compiled version)

marcgpuig, 2018-06-08:
Hi @habibutsu!

Assuming you are running a python client and everithing is connected properly, I think you have a GPU problem. When you are running without `-carla-server` and is not connecting to any client, the simulator only have 1 camera (for visualization). But when you use `-carla-server`, depending which client or configuration are you using, the simulator is using your GPU to render more cameras (sensors) simultaneously. 
This may cause low FPS with old hardware...

habibutsu, 2018-06-08:
@marcgpuig I am not still connected to the server when it already work very slow. For example when i am trying to move corresponded window it to do hard because it responsive very slow. In the same if to open task manager then can see that GPU and CPU not loaded.

Simple script presented below take a time is around `1m25.607s`

```python
import cv2

from carla.client import make_carla_client
from carla.sensor import Camera, Lidar
from carla.settings import CarlaSettings
from carla.tcp import TCPConnectionError
from carla.util import print_over_same_line

host = '192.168.100.6'
port = 2000
player_start = 0

settings = CarlaSettings()

camera0 = Camera('CameraRGB')
camera0.set_image_size(800, 600)
camera0.set_position(0.30, 0, 1.30)
settings.add_sensor(camera0)

def save_image(filename, image):
    if image.max() < 200:
        image = image * 255
    cv_img = cv2.cvtColor(image, cv2.COLOR_RGB2BGR)
    cv2.imwrite(filename, cv_img)

with make_carla_client(host, port) as client:
    print('connected...')

    scene = client.load_settings(settings)
    client.start_episode(player_start)

    client.send_control(
        steer=0,
        throttle=2,
        brake=0.0,
        hand_brake=False,
        reverse=False)

    for frame in range(0, 10):
        print(frame)
        measurements, sensor_data = client.read_data()
        image = sensor_data['CameraRGB'].data

        save_image(f'screen{frame:02}.png', image)
```

marcgpuig, 2018-06-12:
Hi @habibutsu.
I think I can't completely understand you. If the client (the python program) is not connected to the server (the Unreal Engine game), both programs are waiting trying to connect each other so it is normal they are responsively very slow. And that's why the GPU and the CPU are not bussy. So your problem and this _low FPS_ is because the game (simulator) is not _running_ until a client connect to it (when you use `-carla-server`).

closed 2018-06-26 by marcgpuig

Rooparun, 2019-01-25:
Hi, Is there a way to save your camera images in any other format other than png format ?

marcgpuig, 2019-01-25:
@Rooparun 
Your question has nothing to do with this issue. Maybe you want to look first to #842 and #900.
Cheers!