carla-simulator/carla #489: How to make vehicles that are added through Unreal Engine run?

zhangfree2018, 2018-06-11:
Hi , I want to achieve automatic overtaking training without going through the reverse lane. Since the simulator currently does not support multiple lanes, I widened the width of one-way lanes. And added vehicles through the Unreal Engine in the widening lane. But the added vehicle is static. And how to make Non-player agents to run on the extended road?
 I want to know whether the current simulator supports my idea. If it can be realized, what do I need to do?Thanks for your help in advance.

nsubiron, 2018-06-11:
Hi @zhangfree2018,

What you are trying to do is very difficult to achieve with Carla. Our vehicles require a [`URoadMap`](https://github.com/carla-simulator/carla/blob/332d40b58e9d8be23aafe9b47f0fa00ba1642a24/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/RoadMap.h#L90) to drive, this is a sort of image that tells them the direction of the road at any point. Our road generator already generates and serializes this map in editor, but if you use your own roads you'll have to adapt it. Then, the autopilot of the vehicles is very simple and was written with the assumption that there was only one lane, most probably, the vehicles will drive through the center of the widened lane. You probably need to adapt this too.

For the vehicles to start moving you need to spawn their controller too right after spawning the vehicle. The easiest is to put some player starts around and the Carla vehicle spawner should put vehicles with the right controller.

zhangfree2018, 2018-07-05:
Hi @nsubiron ,
Sorry for my delayed reply. Your answer means that I can't modifiy Twon1 and Town2 you provided. How is the image that  tells them the direction of the road at any point obtained? Currently, my demand for maps requires multiple lanes. Because I don't have the experience of building a simulation environment. Can you provid some simple operation documents? I have read the documents about Map customization you provided, but I can't understand.

closed 2018-09-30 by stale[bot]