carla-simulator/carla #492: How to use the benchmark?

NozomiWyane, 2018-06-12:
Ubuntu 16.04 CARLA 0.8.1
I install this version of CARLA,but the doc on the website are the old version of benchmark using guide.Until now I still didn't understand how to run the benchmark and start my own benchmark.please?

felipecode, 2018-06-12:
The benchmark was remodeled on version 0.8.2.
Please use that one. The old one still had issues. 
http://carla.readthedocs.io/en/latest/benchmark_start/

NozomiWyane, 2018-06-12:
emmm, one more question is how to execute this line in source code version CARLA?
./CarlaUE4.sh /Game/Maps/<Town_name> -windowed -world-port=2000  -benchmark -fps=10
thanks!

felipecode, 2018-06-18:
Were you able to solve the issue ?
Which benchmark do you want to use ? Keep in mind that the -benchmark option from Unreal has nothing to do with the driving benchmark in CARLA.

NozomiWyane, 2018-06-26:
Maybe I understood benchmark wrong...In the client.read_data() method,the return value is measurements and sensor_data,I recognized that how to use measurements in carla_server_pb2.py,how to get the other data like tarfficlight or speedlimitsign.Using "measurements.tarfficlight"  is not working.

closed 2018-09-01 by stale[bot]