carla-simulator/carla #498: Lidar to World Transformation

luigui2906, 2018-06-13:
Hi,

First of all, I would like to thank all the CARLA team for the great work done with the simulator. I find it really useful.

I have been working with Lidar data and I am trying to take all the frames received from the sensor at different times to a same coordinate system. I followed the example given in the file point_cloud_example.py where this is done for a pointcloud obtained from the depth map extracted from camera. However, it seems that there are problems with the transformation using the Lidar sensor.

This is how I am defining my sensor:

`if args.lidar:
    lidar = Lidar('VelodyneHDL64')
    lidar.set_position(x=(0.81-0.05), y=0, z=1.73)
    lidar.set_rotation(pitch=0, yaw=0, roll=0)
    lidar.set(
        Channels=64,
        Range=90,
        PointsPerSecond=1200000,
        RotationFrequency=10,
        UpperFovLimit=2,
        LowerFovLimit=-24.8)
    settings.add_sensor(lidar)`

Then I am taking what should be the transformation matrix from the Lidar to the vehicle by doing:
`lidar_to_vehicle_transform = lidar.get_unreal_transform()`

After having this, in a for loop when obtaining every frame point cloud I am obtaining the Vehicle to World transformation and obtaining the Lidar to World by multiplying this two matrices, and finally applying this transformation to the respective pointcloud:
`world_transform = Transform(measurements.player_measurements.transform)
lidar_to_world_transform = world_transform * lidar_to_vehicle_transform
measurement.point_cloud.apply_transform(lidar_to_world_transform)
`
And then the measurement is saved into disk. However if I integrate all frames together, is possible to see that the resulting pointcloud have transformation errors from a frame to the other. I can't see how the coordinate axis of the Lidar and the vehicle are defined in order to calculate and apply the transformation by myself.

Thanks in advance.

minhaoxu1995, 2018-06-14:
#392 this might help.

luigui2906, 2018-06-14:
Thank you Minhao,

That solves the problem. There's still something weird however. When reconstructing the environment with the data from the Lidar everything seems to be inverted, like a mirror. A right turn made by the vehicle seems like a left turn in the reconstructed data. It's not really important, but I would like just to know why this happens.

marcgpuig, 2018-06-14:
@minhaoxu1995 thank you for your answer!

@luigui2906 take a look at Unreal's coordinate system:
[image]
Depending on how you are making the transformation and how you are visualizing the data is normal that you can see a _flip_ on some axis. Could be this the case?

luigui2906, 2018-06-14:
Hi @marcgpuig,

Yep, I think is because the Lidar coordinate system is defined following left-hand rule (YxX gives Z) just as you show in the image. While the vehicle's is defined right-handed (XxY gives Z). That's why we need the `scale(z=-1)` on the unreal transform. And this is where the flip comes from I think.

closed 2018-06-14 by luigui2906

xinshuoweng, 2020-03-12:
can anyone help how to do scale(z=-1) in the lastest Carla version? it seems like that the feature has disappeared since a version.