carla-simulator/carla #503: ubuntu 18.04 : 'xlocale.h' file not found

maxlem, 2018-06-14:
trying to build master

build fails with:
```[ 50%] Building CXX object projects/libcxx/lib/CMakeFiles/cxx.dir/__/src/algorithm.cpp.o
In file included from /.../workspace/carla/Util/Build/llvm-source/projects/libcxx/src/algorithm.cpp:11:
In file included from /.../workspace/carla/Util/Build/llvm-source/projects/libcxx/include/random:1646:
In file included from /.../workspace/carla/Util/Build/llvm-source/projects/libcxx/include/istream:163:
In file included from /.../workspace/carla/Util/Build/llvm-source/projects/libcxx/include/ostream:138:
In file included from /.../workspace/carla/Util/Build/llvm-source/projects/libcxx/include/ios:216:
/.../workspace/carla/Util/Build/llvm-source/projects/libcxx/include/__locale:39:11: fatal error: 
      'xlocale.h' file not found
# include <xlocale.h>
```

see
https://sourceware.org/glibc/wiki/Release/2.26#Removal_of_.27xlocale.h.27

maxlem, 2018-06-15:
I compiled using the proposed workaround
ln -s /usr/include/locale.h /usr/include/xlocale.h

closed 2018-06-18 by marcgpuig

1453042287, 2019-04-03:
@maxlem thanks a lot, it solve my problem :)