carla-simulator/carla #504: Multi lane support and few questions

imran5144, 2018-06-14:
1) Is there multi lane support in the current release ? else when you guys planning to add the multi lane support ?

2) we have requirement to  get vehicles  that are on left and right side of the main car along with the distance to main car, is it possible to get that kind of info form the simulator ?

3)How to change the car form Mustang to a different car ? what other cars are supported ?

p-schulz, 2018-06-15:
Hi @imran5144 

1) I managed to multiple lanes using the splines tool and creating custom roads, highway like with 3 lanes. There you can add some RoutePlanners, every vehicle will follow the route when passing the BoxCollider of a planner.

2) Using the RoutePlanner you can add PlayerStarts in front of them. They will drive through the planners and eventually follow the road. This way you might achieve a behaviour of the Ego Car driving in the middle and two Agent Cars left and right from the Ego. Using the measurements from the server you can dump the locations (Transform) of every vehicle in every frame (look at issue #446 for some simple python client code, you can easily add measurements.non_player_agents for the NPC measurements). 

3) In your CarlaSettings.ini you can look at the following lines to change the Ego Car vehicle. I think every vehicle in carla is supported to work as the Ego vehicle. Personally I use the BMW Gran Tourer - this is from my settings file:
```
[CARLA/LevelSettings]
; Path of the vehicle class to be used for the player. Leave empty for default.
; Paths follow the pattern "/Game/Blueprints/Vehicles/Mustang/Mustang.Mustang_C"
PlayerVehicle=/Game/Blueprints/Vehicles/BmwGrandTourer/BmwGranTourer.BmwGranTourer_C
```

Greetings

nwsweet, 2018-06-25:
@p-schulz, to create the custom roads like you did, did you use the assets already included with Carla, or did you create your own? I am trying to build a bike lane, but don't see any lane markings other than the double yellow in the Carla assets. Thanks for any help!

TheNihilisticRobot, 2018-06-26:
Hi! @nwsweet 

We do not have lanemarkings beyond what is presented in our maps. But you can easily create something similar using simple rectangles combined with our SplineMeshRepeater (Content/Blueprints/SplineMeshRepeater)

sergiocaccamo, 2018-08-28:
Hi @p-schulz , the highway you created seems very good! Did you/Can you share it somewhere? I am having issues modifying the road spline on town01 (v 0.8.4) and it would be nice to see your example.

p-schulz, 2018-10-15:
Sorry again for answering so late...
@z1223343 are you able to drive around yourself in the ego vehicle? The cars should be placed over the bounding boxes of RoutePlanners, so they drop into them when carla starts... If they have no route at all they should be driving in circles, when they are not moving at all something else is going wrong...
@sergiocaccamo @hzyjerry I cannot completely share the roads, but I will try to export them tomorrow... They are somewhat merged together from the carla roads in /Game/Static/Roads (various RoadTiles from there) and my own models as the railing and the road markings (hint, the road marks are just planes with a masked white texture on them). Anyway, I try to package them up as a example ;-)

bjane123, 2018-10-23:
Hi, @p-schulz @TheNihilisticRobot . I am new for calra and UE4.
I don't know how to duplicate  a town01 map correctly. I tried to  copy a town01.umap and change a playerspawner location, then save. But it failed. Could you mind tell me how to duplicate a exist map?

TheNihilisticRobot, 2018-10-26:
Ussualy, when Unreal Doesnt let me Save is because I have two Instances of the same project open. I don't think there is anything wrong with how you duplicated the map.

closed 2019-01-01 by stale[bot]