carla-simulator/carla #505: binocular vision

Archiewyq, 2018-06-15:
How many cameras on the car? 
Can I get two images from two cameras(left camera and right camera, if there is)?

thanks!

p-schulz, 2018-06-15:
Hi @Archiewyq 

you can add cameras using carla settings, either with the CarlaSettings.ini file or directly in your client. This is a stereo setup I use in my python client:
```
                # RGB Sensor left
                camera0 = Camera('Camera_left', PostProcessing='SceneFinal')
                camera0.set_image_size(WINDOW_WIDTH, WINDOW_HEIGHT)
                camera0.set_position(0.30, 0, 1.30)
                settings.add_sensor(camera0)

                # RGB Sensor right
                camera1 = Camera('Camera_right', PostProcessing='SceneFinal')
                camera1.set_image_size(WINDOW_WIDTH, WINDOW_HEIGHT)
                camera1.set_position(0.30, 0.30, 1.30)
                settings.add_sensor(camera1)
```
By setting the position of the individual cameras you can adjust them to be left or right. Define WINDOW_WIDTH and WINDOW_HEIGHT for your desired output or change the values to something like 800 and 600.

marcgpuig, 2018-06-15:
Just like @p-schulz says.

Also there is more issues that with the same question #501 and #227.

closed 2018-06-15 by marcgpuig