carla-simulator/carla #510: adding dynamic objects

Marwa215, 2018-06-18:
Hi, I would like to control the place and time of adding dynamic objects like (signal signs, pedestrians, cars) and add it based on the agent performance. How can I do this...thanks in advance

marcgpuig, 2018-06-18:
Hi @Marwa215.
@nsubiron is working on a new server that will allow you to do this kind of things!
Take a look at #439.

Marwa215, 2018-06-19:
Thanks for your reply, could I know how did you generated the scenarios for town1&2 which files in the simulator?  I tried to run the planner-example which you gave but it gives not permitted. 
My phd project to generate scenarios for targeting some learning objectives and improving the trainee performance level, so I need to inject some actions into the environment at run time such as environmental conditions, pedestrians, traffic signs...etc and update according to the trainee skill level, if this is not available now when the new server may be released.

Kind regards,
Marwa

marcgpuig, 2018-06-19:
@Marwa215, we don't have a release date right now.
Maybe in July you can find some experimental stuff in our repo

doiken23, 2020-01-07:
Hi, thank you for beneficial discussion in this issue.

BTW, can't we spawn signal signs and traffic lights yet? If not, could you tell us when do you plan to release the version that enables that or advise us to implement such functions by ourselves.

Best regards,