carla-simulator/carla #513: 0.8.4 (Linux) - Random crashes when using two-wheeled vehicles

nsubiron, 2018-06-18:
Random crashes occur in PhysX related code concerning the vehicle's wheels. This crash seems to happen only when running the simulation with bikes and motorbikes. The exact place of the crash varies but it's usually due to some NaN value. Possibly some uninitialized memory in the shader that computes the wheel's physics.

Here an extract of the log of one of the crashes: [log.txt](https://github.com/carla-simulator/carla/files/2111938/log.txt).

- - -

As a workaround, for now you can disable all the two-wheeled vehicles with

**CarlaSettings.ini**
```ini
[CARLA/LevelSettings]
; Disable bikes and motorbikes, if true only four-wheeled
; vehicles will be spawned.
DisableTwoWheeledVehicles=true
```

**Python**
```py
settings = CarlaSettings()
settings.set(DisableTwoWheeledVehicles=True)
```

MRAWAY77, 2018-06-19:
is it possible to set bikes or motorbike as player vehicle? i tried using the init file, it cannot load...

felipecode, 2018-07-16:
I observed the same issue also when setting the number of vehicles to 0
but without setting the DisableTwoWheeledVehicles=true.

nsubiron, 2018-11-30:
In 0.9.1, if you are using the `spawn_npc.py` script you can use the `--safe` argument to spawn only vehicles that do not cause accidents. At this point all the bikes are left out, also the Isetta.

    ./spawn_npc.py --safe

germanros1987, 2020-03-19:
This should now be solved by the Traffic Manager.

closed 2020-03-19 by germanros1987