carla-simulator/carla #515: control the traffic signs

lockeregg, 2018-06-20:
I was wondering if I can control the traffic signs?

for example make it stays green

Thank you!

marcgpuig, 2018-06-20:
Hi @lockeregg.
If you set all the traffic lights to green, cars will probably crash...
If you want to do it anyway, @TheNihilisticRobot can explain you how to change it from the editor.

TheNihilisticRobot, 2018-06-21:
Hi @lockeregg 

In each crossing there is an actor called "TraffictLightGroup" that controlls the time every streetlight assigned to it stays in each color however it is programmed to never give pass to more than one entrance at a time. 
If you need a StreetLight that always give pass you may just use the model without any code, you'll find the assets we used to create the lights in the folder Content/Static/TrafficSigns/Streetlights.

lockeregg, 2018-06-22:
Thank you very much!

closed 2018-06-22 by marcgpuig

soumya997, 2022-11-28:
This is a solution. I was stuck in that previously.

```python3
list_actor = world.get_actors()
    for actor_ in list_actor:
        if isinstance(actor_, carla.TrafficLight):
             # for any light, first set the light state, then set time. for yellow it is 
             # carla.TrafficLightState.Yellow and Red it is carla.TrafficLightState.Red
            actor_.set_state(carla.TrafficLightState.Green) 
            actor_.set_green_time(1000.0)
            # actor_.set_green_time(5000.0)
            # actor_.set_yellow_time(1000.0)
```

For more reference see this [carla_dataset_tools](https://github.com/KevinLADLee/carla_dataset_tools/blob/afc43efc9f18472b726261e6a5e9a2aadd3b2c4e/data_recorder.py#L48).