carla-simulator/carla #516: Segmentation for lidar

xslittlegrass, 2018-06-20:
There is a "semantic segmentation" camera that provides the segmentation information for camera images. Is there a similar one for lidar, i.e., how can I get class tags for each lidar point?
Thanks!

nsubiron, 2018-06-21:
Hi @xslittlegrass,

We don't have a Lidar with semantic segmentation tags yet. There has been some attempts on doing that but there are some limitations on how the current lidar implementation works. The main problem is that the ray-casts done by the lidar do not hit the objects but their colliders, and those colliders aren't tagged. See #423 or #455.

closed 2018-07-09 by nsubiron

vishal-prgmr, 2019-05-06:
@nsubiron , the code used for simulating the LIDAR in AirSim (developed by MicroSoft) says that the LIDAR implementation is based on the implementation from CARLA. 

I have managed to label the entire point cloud which i obtain in AirSim. Since the implementations are similar should it not be possible to do the same in CARLA. 

P.S. I still havent started exploring CARLA. Will get back to this thread once I have tried something in CARLA.

vishal-prgmr, 2019-06-06:
@xslittlegrass , @nsubiron , checkout my branch for Semantic labels of the PointCloud. 
[https://github.com/vishal-prgmr/carla/tree/Lidar_PointCloud_Segementation](url)

marcgpuig, 2020-07-31:
This has been added in #3131 :)