carla-simulator/carla #519: How can I change manual_control.py into third-person view?

kanchi0914, 2018-06-20:
Hi.

When I run `manual_control.py`, it starts as first person view game.
However, I want change it into third-person view like carla-server mode.
I mean, I want to debug my environment, so It's fine as long as I have the way to control vehicle manually, while outputting data.
I think `VehicleControl()` is effective but I'm not sure about it.
Is there any way?

marcgpuig, 2018-06-20:
Hey @kanchi0914.
What you can do is just modify the RGB camera and set a custom position and rotation, behind your car.
Something like:
```py
camera0.set_position(-5.0, 0.0, 2.0)
camera0.set_rotation(-20.0, 0.0, 0.0)
```
But if you want to receive the main (_first person_) camera anyway, you can add another camera and set it behind your car, and display the new one while gathering data from your original camera.

kanchi0914, 2018-06-22:
Thank you for your reply. it sounds that will work for me.
So, do you mean We can't apply keyboard input while server-mode without Pygame?

marcgpuig, 2018-06-22:
It is pygame who catch the keyboard events, but you can use another python method to get the input.
What you can't do (right now) is control from the Unreal while using a client.

kanchi0914, 2018-06-23:
OK, I understand. Thank you for your kindness!