carla-simulator/carla #522: Low FPS when run in a server

crazyzxd, 2018-06-21:
Hi, when the CARLA server is run by using the script  CarlaUE4.sh, i enable 2cameras ,capture rgb image, depth image, and i also enable a lidar,  it turns out that i only get a FPS around 3, is this correct？
(My computer has a i7-7700k cpu and a gtx745 gpu.) 
Cause I was using the CARLA  to get images  to train a deep neural networks , the FPS is too low to accept, and i want speed up the process ,how can i get that?

minhaoxu1995, 2018-06-21:
The I/O process makes it run slowly, you can try use `./CarlaUE4.sh -benchmark -fps=10` to get a fixed simulation step. The program will still run slowly but the step between frame will be constant. I think this will help if you are collecting data for training.

crazyzxd, 2018-06-21:
@minhaoxu1995  thanks a lot for your advice ,but it seems doesn‘t  work ,i still get a fps around 3.
Is there any other method?

marcgpuig, 2018-06-22:
3 FPS is a really low framerate...
Which is your framerate when you run CARLA without server-mode?

crazyzxd, 2018-06-23:
@marcgpuig  when i run without server-mode ,i get fps around 10.

marcgpuig, 2018-06-25:
With this information looks like your bottleneck is the GPU.
Without server-mode we can achieve 40 fps with and old gtx 1080 in Linux, and up to 70 fps with a newer gtx 1070 in Windows.

closed 2018-07-03 by marcgpuig