carla-simulator/carla #523: Physics Asset Error

MRAWAY77, 2018-06-21:
Dear admin,

As some of the car model such as BmwGranTourer doesn't come with physics assets so i actually added it in UE4 but however, this appear somehow.... im not sure whats wrong ?

aN/INF in PHit!
[2018.06.21-08.55.31:666][246]LogCollision: Error: RaycastSingle resulted in a NaN/INF in PHit!
[2018.06.21-08.55.31:999][247]LogCollision: Error: RaycastSingle resulted in a NaN/INF in PHit!
[2018.06.21-08.55.32:000][247]LogCollision: Error: RaycastSingle resulted in a NaN/INF in PHit!
[2018.06.21-08.55.32:333][248]LogCollision: Error: RaycastSingle resulted in a NaN/INF in PHit!
[2018.06.21-08.55.32:666][249]LogCollision: Error: RaycastSingle resulted in a NaN/INF in PHit!
[2018.06.21-08.55.32:998][250]LogCollision: Error: RaycastSingle resulted in a NaN/INF in PHit!
[image]
Also, Those cars starts to spin around the terrain in circles!
[image]

MRAWAY77, 2018-06-21:
the physics Assets is the biedge color icon in UE4

TheNihilisticRobot, 2018-06-21:
Hi @MRAWAY77 

That is a problem on our part. There actually is a Physical skeleton for each vehicle we have on the project, but half of them are on the model folder and the other half in the blueprint folder of the vehicle.
We'll optimize our asset folders in the next release to avoid this kind of confusion.

closed 2018-10-27 by stale[bot]