carla-simulator/carla #524: Server Protocol

p-schulz, 2018-06-21:
Hi,
according to the Server Documentation the protocol for the measurements and controls is only write and only read (measurements are written, controls are read). Every message is prepended by 4 bytes uint32 message size. 
My C++ client successfully connects and reads (around 3000 bytes measurements, 19200000 bytes image data)/writes data so far. But I'm a little bit confused about a buffer of length 4 which is sent by the sensor port from the server:

```
[2018.06.21-12.31.34:781][811]LogCarlaServer: Ready to play, notifying client
DEBUG: tcpserver 2000 : sending from buffer of length 6 
DEBUG: tcpserver 2000 : receiving to buffer of length 4 
INFO:  tcpserver 2001 : connected 
INFO:  tcpserver 2002 : connected 
DEBUG: tcpserver 2002 : receiving to buffer of length 4 
[2018.06.21-12.31.34:782][811]LogCarla: Found 23 PlayerStart positions for spawning vehicles
[2018.06.21-12.31.34:887][811]LogSpawn: Warning: SpawnActor failed because of collision at the spawn location [X=-71259.641 Y=77125.609 Z=246.785] for [ChevroletImpala_C]
[2018.06.21-12.31.35:897][811]LogCarla: Spawned all 23 requested vehicles
[2018.06.21-12.31.35:897][811]Cmd: g.TimeoutForBlockOnRenderFence 300000
[2018.06.21-12.31.35:898][811]g.TimeoutForBlockOnRenderFence = "300000"
[2018.06.21-12.31.35:898][811]LogLoad: Took 4.644012 seconds to LoadMap(/Game/Maps/UEDPIE_0_Ulm04)
DEBUG: allocating sensor buffer of 1920032 bytes 
INFO:  allocating agents buffer of 4128 bytes 
DEBUG: tcpserver 2001 : sending from buffer of length 3033 
DEBUG: tcpserver 2001 : sending from buffer of length 1920032 
DEBUG: tcpserver 2001 : sending from buffer of length 4 
DEBUG: tcpserver 2002 : receiving to buffer of length 10 
DEBUG: tcpserver 2002 : receiving to buffer of length 4 
DEBUG: allocating sensor buffer of 1920032 bytes 
INFO:  allocating agents buffer of 4128 bytes 
DEBUG: tcpserver 2001 : sending from buffer of length 3228 
DEBUG: tcpserver 2001 : sending from buffer of length 1920032 
DEBUG: tcpserver 2001 : sending from buffer of length 4 
DEBUG: tcpserver 2002 : receiving to buffer of length 10 
DEBUG: tcpserver 2002 : receiving to buffer of length 4 
DEBUG: tcpserver 2001 : sending from buffer of length 3228 
DEBUG: tcpserver 2001 : sending from buffer of length 1920032 
DEBUG: tcpserver 2001 : sending from buffer of length 4 
```

I think after a few frames the buffers which I'm reading seems to be shifted... What are these 4 bytes?
Also:

Every image is an array of uint32's:
[width, height, type, FOV, color[0], color[1],...]
Is this prepended by int64 framenumber (like in the python client) ?

Thanks in advance :)

nsubiron, 2018-06-21:
:grimacing: you are right, we recently started tagging images with the frame number (uint64) and forgot to update the documentation.

This is where the images are encoded in the server side 
https://github.com/carla-simulator/carla/blob/c4dd6794fc8a2eb32025425cc322d28a578692d3/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureCamera.cpp#L46-L53

and here parsed in the Python client
https://github.com/carla-simulator/carla/blob/c4dd6794fc8a2eb32025425cc322d28a578692d3/PythonClient/carla/client.py#L190-L196

Really sorry about that.

p-schulz, 2018-06-25:
No problem! Thank you ;-)

closed 2018-06-25 by p-schulz