carla-simulator/carla #525: night mode

Archiewyq, 2018-06-22:
hi,
can i add a night-mode for Carla?

thanks

TheNihilisticRobot, 2018-06-22:
hi @Archiewyq 

Sure you can. You just need to create a custom climate in our [weather settings.](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Config/CarlaWeather.ini) Then select it from your [carla config file](http://carla.readthedocs.io/en/latest/carla_settings/). However, we are working in improving our weather system, adding in the process night configuration, so you might want to wait for one of our next releases.

Archiewyq, 2018-06-24:
@TheNihilisticRobot 
thanks for your work！another question: can i do camera calibration with the cameras on the car? Or，can i get the parameters of the cameras?

marcgpuig, 2018-06-25:
Hi @Archiewyq 
Please, next time you want to make a new question open a new issue.

You set your camera parameters on your python client, so there is no need to _get the parameters of the cameras_. If what you want is get the intrinsic and extrinsic matrix from your camera, take a look at [PythonClient/point_cloud_example.py](https://github.com/carla-simulator/carla/blob/master/PythonClient/point_cloud_example.py).
Also take a look at #56, #314.

closed 2018-06-25 by marcgpuig