carla-simulator/carla #526: Aerial view of simulation

sandman, 2018-06-22:
Hi, is there an option for an aerial view of the simulation, from a fixed or varying(ex. Drone) vantage point?

marcgpuig, 2018-06-22:
We do not support this feature right now, but with the future server architecture #439, implement this kind of things should be quite easy!

sandman, 2018-06-22:
Thanks @marcgpuig. I'm wondering if there's a hack that can do the job. How did you generate the videos on the homepage? What if I change the camera coordinates to another point outside the vehicle in world frame? Or define a new camera at a fixed position in world frame..?

TheNihilisticRobot, 2018-06-26:
Hi @sandman 

Our videos are done using the vehicle camera for shots that follow the vehicle and UE4 Native cameras that you can move around the level setting some keypoints but do not have option to take Depth or semantic segmentation. Here is the documentation for unreal's cinematic system: [Sequencer Editor](https://docs.unrealengine.com/en-us/Engine/Sequencer/Overview)

closed 2018-07-25 by TheNihilisticRobot

eeedWard, 2019-11-26:
Hi @sandman . Did you find any solution or workaround to get an aerial view? I am also interested in this!