carla-simulator/carla #532: Destroying NPC in Unreal Engine?

anthonypan08, 2018-06-25:
I am curious about how to destroy an NPC in Unreal Engine (when it hits a trigger box for example).

[image]

This is what I have currently but whenever Unreal Engine tries to destroy the NPC the entire editor shutdown.

marcgpuig, 2018-06-25:
Hi @anthonypan08.
Try to destroy the controller before the actor.

anthonypan08, 2018-06-25:
How would I do that? I tried looking for the destroy controller action in blueprint but couldn't find it

marcgpuig, 2018-06-27:
Sorry for the late response!
Have you tried [this](https://answers.unrealengine.com/questions/557499/ai-controller-can-never-be-destroyed-or-be-garbage.html) solution from Unreal Forums?

closed 2018-09-02 by stale[bot]