carla-simulator/carla #533: Creating NPC using Vehicle Spawner?

anthonypan08, 2018-06-25:
I notice that the player spawner uses the custom defined class vehicle spawner. Therefore, I wonder if it is possible for me to somehow spawn a vehicle during a certain event (when I hit a trigger box for example) as opposed to spawning all the vehicles at the beginning.

nsubiron, 2018-06-26:
Hi @anthonypan08,

If you want to do it in C++, the easiest is modifying the VehicleSpawnerBase.cpp. You can add trigger boxes in C++ too.

If you want to do it with Blueprints, you can use SpawnActor to spawn the vehicle, and immediately after you should spawn its controller (SpawnDefaultController). If you want to use our WheeledVehicleAIController for the autopilot, you also need to provide a RoadMap. The road map can be retrieved from the CarlaMapGenerator (the road generator), present in the level.

Basically you would need to replicate the following code 
https://github.com/carla-simulator/carla/blob/21969778798581d189b84a477199c1bb1417f615/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleSpawnerBase.cpp#L100-L119

anthonypan08, 2018-06-26:
[image]

Here is what I have currently. However, there are 2 issues

1) The spawned vehicle is static
2) It spawns at the wrong location (it seems to spawn in a player start location for some reason)

anthonypan08, 2018-07-05:
Any updates or similar problems?

nsubiron, 2018-07-05:
Ok, I see. You should not spawn a Carla Map Generator, this is already present in the map and there should be only one. You should search for it at begin play, then retrieve the road map from it, and pass it to the newly created vehicle. AFAIK the controller won't work like that either, I think you should either use the spawn default controller function (as in the C++ snippet), or call Possess after spawn.

anthonypan08, 2018-07-06:
How exactly do I "pass the road map" to the vehicle? Should I promote the vehicle to a variable or something?

nsubiron, 2018-07-06:
Ouch you are right, I thought that function was exposed to blueprints but it's only available on C++, sorry about that. However, you can expose it by adding one line to our code. Just add `UFUNCTION(Category = "Road Map", BlueprintCallable)` on top of the function

```diff
--- a/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.h
+++ b/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.h
@@ -83,6 +83,7 @@ public:
   /// @{
 public:
 
+  UFUNCTION(Category = "Road Map", BlueprintCallable)
   void SetRoadMap(URoadMap *InRoadMap)
   {
     RoadMap = InRoadMap;
```

After recompiling you can call that function from blueprints. From the "Return Value" of the Spawn Vehicle node you should be able to get the "Set Road Map" function. The Carla Map Generator has a "Get Road Map" function.

anthonypan08, 2018-07-09:
[image]

Thanks @nsubiron so much for you help. Here is a screenshot of my blueprint for any future references

closed 2018-07-09 by anthonypan08

gaurish123, 2019-02-26:
@anthonypan08 -

Hi - Need small pointer ...

1. I want to add my "Auto"- vehicle model in carla simulation environment. I just have .fbx vehicle file how can I proceed further . when I am trying to import its .fbx as Skeletal Mesh, selecting FordMustangGt1966_skeleton (which is 4 wheeled) , it gives error as it is expecting skeleton for 3- wheeled vehicle. Any Idea how can I achieve this.

2. Once This is done can I make use of above blueprint to spawn my auto vehicle as npc in carla simulation environment?

Thanks