carla-simulator/carla #540: Real World Maps using Infraworks

abol-karimi, 2018-06-27:
Hi,

I'm doing research on safety of AVs and I need real world 3d maps for the UE4 level. I've been web surfing for solutions for quite a while. Autodesk's Infraworks can easily connect to open databases such as OpenStreetMap and automatically build a 3d model that includes roads, buildings and ground imagery. Then you can export the model to an fbx format and import it in UE4. It would be great if the roadmap could be generated according to the roads in the fbx file.

Best,
Abol

p-schulz, 2018-06-28:
Hi @abol-karimi 
if you're looking for integrating real world road networks, there is already a plugin by one of the Epic Games developers that supports the generation of roads and buildings: https://github.com/ue4plugins/StreetMap
IMHO a deeper integration of realworld roads is difficult at the moment, but might be easier based on xml/osm data than fbx files, as they are text files for meshes. Still, creating real life like roads from open databases is a task that requires some work done by yourself. Curves or intersections are stored as multiple segments, I for myself use the road network as a base for dragging splines along the roads and attaching RoutePlanners. 
Maybe you can somehow make use of the plugin. But I'd like to see open database integration too. :)

marcgpuig, 2018-06-28:
Hello @abol-karimi @p-schulz.
This is a feature that we absolutely want to incorporate in CARLA, but is not a easy task.

We are thinking on a procedural way to generate all these meshes, paths, buildings and so on, so you can use an easy tool to manually or automatically create all the roads you want. There is a lot of things to considere here, so as mentioned before, this is not trivial and we are slowly but carefully working on it.

Right now you can import models from CityEngine to Unreal Engine. It's a little bit tedious and we have no automated way to generate the paths for vehicles and pedestrians but... it's something.

Here you have a work in progress sample that @XGodina is working on, using CityEngine to generate the roads from OpenStreetMaps and our assets for buildings and so on.

[image]

mvpcom, 2018-07-06:
This is a great idea, and it is a fantastic and futuristic feature for the Carla. I believe it is vital for the self-driving car research.

closed 2018-09-11 by stale[bot]

simon0628, 2019-03-07:
> no automated way to generate the paths for vehicles and pedestrians but... it's something.

Hi! Thanks for contributing. This is exactly what I need. 👍 
I wonder if there's any notable progress about this work or will you add such feature in future release?