carla-simulator/carla #541: World height of Town 01 has incorrect scale

danil-tolkachev, 2018-06-28:
It seems that simulator sends height in centimeters for Town01.

According to `Town01.txt` world shift for Z axis is `-0.3811` (in meters, I suppose), and for `Town02.txt` it is `-0.22`. For Town02 the actual height of the objects is the same as in the corresponding text file (without minus sign), but for Town01 all objects are located about 38 meters above the ground.

Related to #537.

felipecode, 2018-06-28:
I changed to 0.38 in the files after we changed the reference to meters.
Are objects really located at 38 meters above the ground  @nsubiron  @TheNihilisticRobot  ?

TheNihilisticRobot, 2018-06-28:
That is true. The road in  Town01 is 38 meters above the origin of the world (3811.0 cm above, exactly )

marcgpuig, 2018-07-03:
Hey @danil-tolkachev.
Did these answers solved your question?

felipecode, 2018-07-04:
I also think is inconsistent. I will try to solve it and I will check your PR @danil-tolkachev 
Thanks a lot !

danil-tolkachev, 2018-07-04:
@felipecode Oh, my PR is only related to this issue and not fixes it.

nsubiron, 2019-04-05:
This was corrected.

closed 2019-04-05 by nsubiron