carla-simulator/carla #544: Traffic light and speed limit taking effect

petroolg, 2018-06-28:
Hi,
is there a way how to get a traffic light and speed limit taking effect? This information is printed in simulator window (4th and 5th line). 

[image]

It can be done using measurements API http://carla.readthedocs.io/en/latest/measurements/ by determining the nearest traffic light/speed sign, but maybe there's a simpler solution I'n not aware of. 

Thanks

marcgpuig, 2018-06-29:
Hi @petroolg.
Sorry but this is the only way right now :(
We are currently changing the server so we will keep this in mind.

closed 2018-09-04 by stale[bot]

kbarbora, 2019-08-06:
Does anyone know how to get the speed limit and traffic light status as in the image for version 9.6?
Thanks for your help