carla-simulator/carla #545: Crash when runing client_example.py on windows

imran514, 2018-06-28:
I am trying to run carla simulator in client-server mode  and when i run the python client, i get below crash.

C:\Carla\_CARLA_0.8.4\0.8.4\PythonClient>python client_example.py
INFO: listening to server localhost:2000
CarlaClient connected
Starting new episode at 'Town01'...
Traceback (most recent call last):
  File "C:\Carla\_CARLA_0.8.4\0.8.4\PythonClient\carla\client.py", line 174, in _read_sensor_data
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "client_example.py", line 248, in <module>
    main()
  File "client_example.py", line 235, in main
    run_carla_client(args)
  File "client_example.py", line 113, in run_carla_client
    measurements, sensor_data = client.read_data()
  File "C:\Carla\_CARLA_0.8.4\0.8.4\PythonClient\carla\client.py", line 127, in read_data
    return pb_message, dict(x for x in self._read_sensor_data())
  File "C:\Carla\_CARLA_0.8.4\0.8.4\PythonClient\carla\client.py", line 127, in <genexpr>
    return pb_message, dict(x for x in self._read_sensor_data())
RuntimeError: generator raised StopIteration

marcgpuig, 2018-07-02:
Hello @imran514.
Did you modified the client?

imran5144, 2018-07-02:
@marcgpuig no i didn't modify the code

marcgpuig, 2018-07-03:
@imran5144,
Right now we support Python **2.7**, **3.5** and **3.6**.
It looks like you are using Python 3.7 and we didn't know this [change](https://www.python.org/dev/peps/pep-0479) on 3.7 version.

Thanks for your feedback! We'll keep this in mind.

imran514, 2018-07-05:
Thank you @marcgpuig , it works on 3.6.

closed 2018-07-06 by marcgpuig