carla-simulator/carla #547: The client.start_episode(start_index) function takes too much time which is bad for training RL algorithms.

tjuHaoXiaotian, 2018-07-01:
Hello, I tried this reset function (start_episode(start_index)) and found that it almost takes 30 seconds. It's really bad for training RL algorithms. And my server env is Win10 (with a graphics card GTX 1060). 

Here is the simple code and the result. Is there anything wrong?
```
                    print('Starting new episode...')
                    started_time = time.time()
                    client.start_episode(start_index)
                    print("started in {} seconds...".format(time.time() - started_time))
```
```
Starting new episode...
started in 30.860573768615723 seconds...
```

felipecode, 2018-07-01:
Are you running without display and selecting GPU ?

ambling, 2018-07-15:
Hi, I am facing the same problem on a Ubuntu 16.04 system (32G RAM, GTX1080): a new episode usually takes more than 15 seconds to load the map.

> [2018.07.15-07.11.52:518][675]LogCarlaServer: Received new episode
> [2018.07.15-07.11.52:518][675]LogCarla: Loading CARLA settings from string
> [2018.07.15-07.11.52:518][675]LogCarlaServer: Restarting the level...
> [2018.07.15-07.11.52:523][676]LogNet: Browse: /Game/Maps/Town01?Name=Player?restart
> [2018.07.15-07.11.52:555][676]LogLoad: LoadMap: /Game/Maps/Town01?Name=Player
> [2018.07.15-07.12.05:620][676]LogAIModule: Creating AISystem for world Town01
> [2018.07.15-07.12.09:615][676]LogCarlaServer: Ready to play, notifying client
> [2018.07.15-07.12.09:617][676]LogCarla: Found 152 PlayerStart positions for spawning vehicles
> [2018.07.15-07.12.09:620][676]LogLoad: Took 17.065060 seconds to LoadMap(/Game/Maps/Town01)

I have tried to run without display, it seems that the server needs to reload map for each episode as well, and the time is not saved. It seems that a call of "RestartLevel" on the server would always lead to an expensive reloading process. Correct me if I am wrong.

I notice that AirSim has a ["reset"](https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#common-apis) API that moves the car to its start position efficiently. Is there any way to realize that in Carla?

felipecode, 2018-07-16:
That waiting time is not common. Which hardware are you using ? 
I have noticed very big delays on resets when using SDL to select GPU.
But depending on your hardware 15 seconds is common.
We don't just move the objects since the idea of reset is also to recover the initial states of everything.
The reset also gives you flexibility to reload the weather and soon also the map. Maybe you could open an issue so we add a new feature like RepositionAgents ?

ambling, 2018-07-18:
@felipecode Thanks for your response and help.

I find that the delay of loading occurs only when I start the game from UE4Editor:

> UnrealEngine-4.18/Engine/Binaries/Linux/UE4Editor carla/Unreal/CarlaUE4/CarlaUE4.uproject -carla-server -benchmark -game -windowed -ResX=1024 -ResY=720

But if I launch the game via the **CarlaUE4.sh** script in the distribution package, the loading time would be reduced to about 2 seconds. So it may not relate to the hardware.

I do not understand the differences between these two launch methods. If it is because of my false usage, I must say sorry for that, and maybe some sentences in the docs are very helpful.

For the feature of `RepositionAgents`, I wonder if the new client API (#439 ) can be used directly, just to discard the old agent and to start a new agent at the start point, without restarting the game.

closed 2018-09-23 by stale[bot]